DDorch/ScadaSupAgro #22: NetDataLogger: Stop fonctionning after reseting arduinos

DDorch, 2019-07-04:
Steps to reproduce:
- Reboot the RPI
- Switch on the flowmeters and arduinos connected on them
- Data are correctly received by the RPI
- Switch off and on the arduino 
- Data are not received by the arduino whereas arduinos seems to get a correct response 

However, this error does'nt seemed to be systematic.

DDorch, 2019-07-11:
A weird message in /var/log/messages seems to be related to the disfunction:

` TCP: request_sock_TCP: Possible SYN flooding on port 34060. Sending cookies.  Check SNMP counters.`

DDorch, 2019-07-11:
On arrive à obtenir une info de log relative au problème de connection non traitée:

```
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    536 SYNs to LISTEN sockets dropped
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    767 SYNs to LISTEN sockets dropped
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    883 SYNs to LISTEN sockets dropped
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    917 SYNs to LISTEN sockets dropped
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    947 SYNs to LISTEN sockets dropped
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    984 SYNs to LISTEN sockets dropped
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    1016 SYNs to LISTEN sockets dropped
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    1081 SYNs to LISTEN sockets dropped
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    1123 SYNs to LISTEN sockets dropped
pi@raspberrypi:/etc/sysctl.d $ netstat -s | grep "SYNs to LISTEN"
    1159 SYNs to LISTEN sockets dropped
```

D'après https://serverfault.com/questions/482277/deactivate-syn-flooding-mechanism/482279, le paramètre net.ipv4.tcp_max_syn_backlog de sysctl pourrait améliorer la situation:

J'ai tenté
```
pi@raspberrypi:/etc/sysctl.d $ sudo sysctl -w net.ipv4.tcp_max_syn_backlog=5000
net.ipv4.tcp_max_syn_backlog = 5000
```
Et le plantage a été immédiat.
J'ai ensuite tenté avec la valeur 65536 sans plus de succès.

La modification du paramètre `net.ipv4.tcp_syncookies=0` semble résoudre le problème.

Le fichier de conf ajouté avec ces paramètres est `/etc/sysctl.d/local.conf`

guiguitt, 2019-07-11:
le scada à de nouveau planté à 13:51:22 environ.